Xinyuan-LilyGO/T-CAN485 #1: Wrong URL in Reame for Arduino Board install

milen100, 2022-01-23:
URL should be fixed to : https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json .
In the REAME now there is missing slash which cases URL access to fail when the URL is added in Arduino Preferences page.

Roman-Jordan, 2022-01-23:
Good call on this, It would be nice to identify procedures to load ones software on the module. 
  * Press and hold `reset` and `boot` button. 
  * Release the `Reset` button, and then the `boot` button.

LilyGO, 2022-01-24:
Thank you for your suggestion

closed 2022-01-24 by LilyGO

Roman-Jordan, 2022-01-28:
Are you using PlatformIO? I use the esp32 one..

crow1170, 2022-01-28:
I'm trying to use the Arduino IDE, and while an Esp32 Dev board exists, it's throwing an error in the compile step:

> In file included from c:\$$$\documents\arduinodata\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2\xtensa-esp32-elf\include\c++\8.4.0\memory:63,
                 from C:\$$$\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.2\cores\esp32\Esp.cpp:24:
c:\$$$\documents\arduinodata\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2\xtensa-esp32-elf\include\c++\8.4.0\bits\allocator.h:46:10: fatal error: bits/c++allocator.h: No such file or directory
 #include <bits/c++allocator.h> // Define the base class to std::allocator.
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board ESP32 Dev Module.

I expected the Boards Manager would have prevented such an error, but I'm not even sure that everything would work if that error is resolved.

The board doesn't behave any differently when I perform the reset/boot pattern you mentioned, except to send some text over the Serial Monitor:

> rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download