Xinyuan-LilyGO/T-CAN485 #2: modbus rs485/can to mqtt software on T-CAN485

xjustbmw, 2022-02-05:
Hi there,

I purchased the t-can485 board for the following project:

I'm trying to achieve rs485/canbus to mqtt converter over wifi ( and optional if there is the possibility to add rj45 wired lan extension to it)

Anybody can help me with this software ? Should be similar to Friendtrol's GW-1205B interface.
Read/write a few registers over mqtt protocol. Config the initial wifi credentials in AP mode.

Im interested in buying at least 200 boards like this if the software works.
I can also support this software development if necessary.

Many thanks,
Adrian

Roman-Jordan, 2022-02-05:
That does not sound to difficult 🤔 .. Fork and drop me an invite.

Roman-Jordan, 2022-02-06:
As this does not seem to be an issue with the board. Would you mind closing this and opening it up in a fork?

closed 2022-02-06 by xjustbmw