Xinyuan-LilyGO/T-CAN485 #3: Arduino IDE set up

antoineTK, 2022-02-07:
Hi team!

I wanted to make sure I was doing the right things setting the Arduino IDE

1- As described in readme file: add https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json

2- Restart IDE

3- In board manager look up esp32 and download latest version of esp32 by Espressif Systems

4- When building and uploading choose "ESP32 Arduino --> ESP32-Dev Module"

antoineTK, 2022-02-07:
If I do it this way and I try building "Freenove_WS2812_Lib_for_ESP32" example sketches I get

In file included from C:\Users\Antoine\Documents\Arduino\libraries\Freenove_WS2812_Lib_for_ESP32\examples\Rainbow\Rainbow.ino:1:
C:\Users\Antoine\Documents\Arduino\libraries\Freenove_WS2812_Lib_for_ESP32\src/Freenove_WS2812_Lib_for_ESP32.h:35: warning: "RMT_DEFAULT_CONFIG_TX" redefined
 #define RMT_DEFAULT_CONFIG_TX(gpio, channel_id)      \
 
In file included from C:\Users\Antoine\Documents\Arduino\libraries\Freenove_WS2812_Lib_for_ESP32\src/Freenove_WS2812_Lib_for_ESP32.h:18,
                 from C:\Users\Antoine\Documents\Arduino\libraries\Freenove_WS2812_Lib_for_ESP32\examples\Rainbow\Rainbow.ino:1:
C:\Users\Antoine\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32/include/driver/include/driver/rmt.h:87: note: this is the location of the previous definition
 #define RMT_DEFAULT_CONFIG_TX(gpio, channel_id)      \
 
In file included from C:\Users\Antoine\Documents\Arduino\libraries\Freenove_WS2812_Lib_for_ESP32\src\Freenove_WS2812_Lib_for_ESP32.cpp:12:
C:\Users\Antoine\Documents\Arduino\libraries\Freenove_WS2812_Lib_for_ESP32\src\Freenove_WS2812_Lib_for_ESP32.h:35: warning: "RMT_DEFAULT_CONFIG_TX" redefined
 #define RMT_DEFAULT_CONFIG_TX(gpio, channel_id)      \
 
In file included from C:\Users\Antoine\Documents\Arduino\libraries\Freenove_WS2812_Lib_for_ESP32\src\Freenove_WS2812_Lib_for_ESP32.h:18,
                 from C:\Users\Antoine\Documents\Arduino\libraries\Freenove_WS2812_Lib_for_ESP32\src\Freenove_WS2812_Lib_for_ESP32.cpp:12:
C:\Users\Antoine\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32/include/driver/include/driver/rmt.h:87: note: this is the location of the previous definition
 #define RMT_DEFAULT_CONFIG_TX(gpio, channel_id)      \
 
In file included from C:\Users\Antoine\Documents\Arduino\libraries\Freenove_WS2812_Lib_for_ESP32\src\Freenove_WS2812_Lib_for_ESP32.cpp:12:
C:\Users\Antoine\Documents\Arduino\libraries\Freenove_WS2812_Lib_for_ESP32\src\Freenove_WS2812_Lib_for_ESP32.cpp: In member function 'bool Freenove_ESP32_WS2812::begin()':
C:\Users\Antoine\Documents\Arduino\libraries\Freenove_WS2812_Lib_for_ESP32\src\Freenove_WS2812_Lib_for_ESP32.h:53:5: error: no match for 'operator=' (operand types are 'rmt_config_t' and '<brace-enclosed initializer list>')
     }
     ^
C:\Users\Antoine\Documents\Arduino\libraries\Freenove_WS2812_Lib_for_ESP32\src\Freenove_WS2812_Lib_for_ESP32.cpp:26:11: note: in expansion of macro 'RMT_DEFAULT_CONFIG_TX'
  config = RMT_DEFAULT_CONFIG_TX(pin, rmt_chn);
           ^~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\Antoine\Documents\Arduino\libraries\Freenove_WS2812_Lib_for_ESP32\src\Freenove_WS2812_Lib_for_ESP32.h:18,
                 from C:\Users\Antoine\Documents\Arduino\libraries\Freenove_WS2812_Lib_for_ESP32\src\Freenove_WS2812_Lib_for_ESP32.cpp:12:
C:\Users\Antoine\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32/include/driver/include/driver/rmt.h:81:3: note: candidate: 'rmt_config_t& rmt_config_t::operator=(const rmt_config_t&)'
 } rmt_config_t;
   ^~~~~~~~~~~~
C:\Users\Antoine\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32/include/driver/include/driver/rmt.h:81:3: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to 'const rmt_config_t&'
C:\Users\Antoine\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32/include/driver/include/driver/rmt.h:81:3: note: candidate: 'rmt_config_t& rmt_config_t::operator=(rmt_config_t&&)'
C:\Users\Antoine\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32/include/driver/include/driver/rmt.h:81:3: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to 'rmt_config_t&&'
exit status 1
Erreur de compilation pour la carte ESP32 Dev Module

Roman-Jordan, 2022-02-07:
> note: no known conversion for argument 1 from '' to 'const rmt_config_t&'
> note: no known conversion for argument 1 from '' to 'rmt_config_t&&'

I have no idea at this moment but might be able to poke at it a bit later
today. Gotta get in the line for the homeless shelter early if a want a
place to stay.. so.. we shall see if I get time. Hahaha, your only valuable
in america if you do drugs and have kids. But if you spend your time
learning skills that no one understands.. you deserve to be homeless cause
your less than poop.

Roman-Jordan, 2022-02-07:
Sorry, just woke up when I had seen this. I will have some time this
evening sometime to poke at this if you want.

antoineTK, 2022-02-07:
No worries man, I hope your day is going ok!

I was thinking I might use FastLED if I can not figure this out

Roman-Jordan, 2022-02-07:
Nice. If you got time for a call. I can walk you through it.

Roman-Jordan, 2022-02-07:
42427257fourTwo Drop me a text when your ready.

LilyGO, 2022-02-09:
Hello, it can be seen from the log that the WS2812 library is not installed;
1) First compile the flash program from the example to test whether the basic environment is completed.
2) Install all required dependencies
- https://github.com/Freenove/Freenove_WS2812_Lib_for_ESP32
- https://github.com/miwagner/ESP32-Arduino-CAN

closed 2022-02-09 by LilyGO

antoineTK, 2022-02-09:
The library was installed but I was getting it from the T-CAN485 repo. Getting it from the repo you linked worked for me, thank you!

In the board selection, can you confirm the default parameters are correct?

[image]

antoineTK, 2022-02-09:
[image]

LilyGO, 2022-02-10:
There is no problem with your parameter settings.

LilyGO, 2022-02-10:
FLASH MODE please select DIO

antoineTK, 2022-02-14:
@LilyGO any ideas?