Xinyuan-LilyGO/T-CAN485 #4: CAN bus continuously publishes and blocks the programme

Raul-Riera, 2022-05-10:
Hello, good morning!

I have a problem with the CAN bus.
I'm trying the example code found in example/Arduino/CAN
Although the bus should publish at 1000ms intervals, the bus publishes continuously and blocks the program. It never reaches line 89.
Can you help me? Do you know what is wrong?

Thank you very much for your help.

LilyGO, 2022-09-05:
There may be failures caused by unsuccessful sending. It is recommended that you try connecting the cable before powering on.

dalathegreat, 2023-04-02:
For anyone still having issues with CAN, I have a repository set up here that uses both CAN and Modbus RTU at the same time on the LilyGo Hardware: https://github.com/dalathegreat/BYD-Battery-Emulator-For-Gen24

closed 2024-07-10 by Llgok