Xinyuan-LilyGO/T-CAN485 #5: Tutorial

hanusek, 2022-05-18:
Hello. 
I bought LILYGO ESP32 CAN RS485 and take it from the box. I would like to upload to it an example program but I haven't any information. I need a basic tutorial.

LilyGO, 2022-06-27:
1. Install ESP32 board support
2. Connect the USB, select the port
3. Click Upload

closed 2022-06-27 by LilyGO

newdamage1, 2024-07-20:
It is more complicated than three steps. The Arduino IDE asks for the board; This board is not listed as an option, esp32 selections error about missing files.

We don't expect a step-by-step for beginners, but your effort at decent documentation would be appreciated by those who spend money on your products.

danishzd, 2025-07-08:
I am having trouble setting up one that I just bought, My computer does not detect any hardware connection and the board has rapidly switching lights when its plugged in. What am i missing?

lewisxhe, 2025-07-09:
Is the CH9102 driver installed? You should be able to see a COM device in the Windows Device Manager.

lewisxhe, 2025-07-09:
Is it convenient to shoot a video?
, the steps are as follows:
1. Open the computer device manager
2. Insert the USB
3. Check if there is a device in the Port list

lewisxhe, 2025-07-09:
Device Manager doesn't even respond? Refreshing the device doesn't work?

danishzd, 2025-07-09:
No, strangely it doesn’t do a thing, not even a sound

lewisxhe, 2025-07-10:
OK. Then it is damaged. Normally it will recognize the USB and see the USB device in the device manager. If you have just purchased it, please contact the dealer for after-sales service!

lewisxhe, 2025-07-15:
I have received your video and I have already talked to them. They should have arranged the shipment for you.

Almendronman, 2026-01-21:
> I am having trouble setting up one that I just bought, My computer does not detect any hardware connection and the board has rapidly switching lights when its plugged in. What am i missing?

Same here. My experience is:
1. If I connect the board to a PC's port, nothing happens. Nothing in Linux neither Windows (and yes, I have the drivers)
1.1 In Linux, I can not see any /dev/ttyUSBx neither /dev/ttyACMx
1.2 In Windows, the 'Device Manager' doesn't even blink.
2. If I connect the board to a USB-Smart Charger, nothing happens.
3. If I connect the board to a 'dumb' USB Charger, then the LED starts working and changing colors. The PCB gets really hot, and after a while it stops the LED till I press the Reset or Boot button.