Xinyuan-LilyGO/T-CAN485 #16: RS485 ESPHome connect issue

Yocee84, 2023-12-03:
Hi! I cant use this panel. Please share correct esphome setup. 

I use this:

substitutions:
  devicename: *******
  description: ********
  entity_prefix: *******

esphome:
  name: "${devicename}"
  comment: "${description}"

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  level: DEBUG
  baud_rate: 0

web_server:
  port: 80

# Enable Home Assistant API
api:
  encryption:
    key: "*****************************************"
    
ota:

wifi:
  networks:
  - ssid: !secret wifi_ssid
    password: !secret wifi_password
  - ssid: *********
    password: ************
  reboot_timeout: 0.05h

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "*****2"
    password: "**********"

captive_portal:

light:
  - platform: neopixelbus
    type: GRB
    variant: WS2812
    pin: GPIO04
    num_leds: 1
    name: "Light"

uart:
  id: mod_bus
  tx_pin: 22
  rx_pin: 21
  baud_rate: 9600
  stop_bits: 1

modbus:
  flow_control_pin: 5
  id: heatpump_modbus

Bwooce, 2023-12-09:
Try this for some inspiration, I'm working on a homeassistant forum post to go with it. This is working for my application.

https://gist.github.com/Bwooce/7c43622dcc9bc1fbb04d7a0f1e5528df

Yocee84, 2023-12-12:
> Try this for some inspiration, I'm working on a homeassistant forum post to go with it. This is working for my application.
> 
> https://gist.github.com/Bwooce/7c43622dcc9bc1fbb04d7a0f1e5528df

That's how it works for me. Thank you very much!
You should get a commission. Is there a buy me a coffee link?

Saentist, 2023-12-29:
This is how I make it to work in ESPHome with Growatt inverter

`Output` section is important obviously, without it it not work.

```yaml
# Set pins required for LilyGo T-CAN485 board
output:
  - platform: gpio
    id: ENABLE_PIN # Enable the chip
    pin:
      number: GPIO19
      inverted: true
  - platform: gpio
    id: SE_PIN # Enable autodirection
    pin:
      number: GPIO17
      inverted: true
  - platform: gpio
    id: ENABLE_5V_PIN # Enable 5V pin for RS485 chip
    pin:
      number: GPIO16
      inverted: true

# UART MODBUS configuration entry
uart:
  id: mod_uart
  rx_pin: GPIO21
  tx_pin: GPIO22
  baud_rate: 9600 # The out-of-box baud rate for my particular sensor->RS485 bridge
  stop_bits: 1
  parity: NONE
#  data_bits: 8
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: "\n"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);

modbus:
  id: modbus1
  uart_id: mod_uart
#  flow_control_pin: GPIO4

modbus_controller:
  - id: Growatt_MIN600TLX
    address: 0x1   ## address of the ModBUS slave device on the bus
    modbus_id: modbus1
    setup_priority: -10
```

drawlusyk, 2024-05-22:
[Archive.zip](https://github.com/Xinyuan-LilyGO/T-CAN485/files/15409245/Archive.zip)

Here are my can1 and can2 files. Connect h to h and l to l. Sends test data back and forth.

closed 2024-07-10 by Llgok

dvbit, 2024-11-01:
Just. Ought one device. I have a doubt about the connections. As far as I know a b and gnd are needed.in the board Iconnector I see just an and b on pin 20 and 21 . Where is gnd?

Sleeper85, 2025-01-11:
> Here are my can1 and can2 files. Connect h to h and l to l. Sends test data back and forth.

Thanks for sharing your examples, so you confirmed that it is necessary to enable (set to high) PIN 23 to make the CAN chip work ?

Llgok, 2025-01-13:
@Sleeper85 
For using CAN, you can refer to this example：[CAN](https://github.com/Xinyuan-LilyGO/T-CAN485/blob/arduino-esp32-libs_v3.0.1/examples/CAN/CAN.ino)
[image]
[image]

Sleeper85, 2025-01-16:
@Llgok 

Thanks this info really helped me and it works fine with ESPHome and these settings for RS485 and CAN.

```YAML
# +--------------------------------------+
# | LilyGo T-CAN485 related config       |
# +--------------------------------------+

output:
  # MAX13487E SHDN (enable RS485 chip)
  - platform: gpio
    id: RS485_SE
    pin:
      number: 19
      inverted: true # set HIGH level
  # MAX13487E RE (enable RS485 autodirection)
  - platform: gpio
    id: RS485_EN
    pin:
      number: 17
      inverted: true # set HIGH level
  # RS485 and CAN Boost power supply
  - platform: gpio
    id: CGQ_EN
    pin:
      number: 16
      inverted: true # set HIGH level
  # SN65HVD231DR RS PIN (CAN high speed mode)
  - platform: gpio
    id: CAN_SE
    pin:
      number: 23
      inverted: false # set LOW level
```