Xinyuan-LilyGO/T-CAN485 #19: Unusual voltage levels on RS485

bobemoe, 2024-01-30:
# Issue

We've been using these lilygo successfully for a while now using both CAN and RS232 IO and have working firmware communicating with other devices on the RS485 bus.

We've just invested in a load more of these but cant get any of them communicate using the known working firmware on our known working RS485 bus.

To diagnose we hooked up the A and B lines to the oscilloscope, and observe unusual voltage levels.

# Expected Behaviour:

This is taken from other devices communicating on the modbus, as expected, when idle both lines are near 0. When data they are inverted square signal of each other going up to about 3.3v.

[image]
[image]

# Actual Behaviour:

This is taken from the output of the lilygo. Seems the idle state voltage levels are high (both at 3.3V), and during data transmission, one line exhibits a voltage level that is not typical for RS-485 standards going up, it seems offset.

[image]
[image]

# Steps to Reproduce:

Programme lilygo to output data on the RS232 and monitor the voltage levels on both lines during idle and transmission states using a oscilloscope.

# Additional Info

Unfortunately all our original working lillygo are out in the field so we can't make a direct comparison of these ones to our known working ones.

If we connect both lines to the logic analizer then we only see data on one line, as the yellow trace above is just detected as high. I know during development with the original boards we used the logic analizer extensively and both lines were detected.

Has something changed? We've tried adding/changing termination and biasing resistors with no luck.

We have 25 of these and have tested about 5 so far all exhibiting same issue. I believe they are genuine units. Can we provide batch number or further details?

lewisxhe, 2024-02-03:
Hi @bobemoe 

Can you use two T-CAN485 for transceiver loopback testing?
First check whether the hardware can work normally?
I noticed that you said the idle voltage is about 3V. Did you measure this by connecting it to another bus or not?

bobemoe, 2024-02-03:
@lewisxhe Thanks for your response. I am in the workshop Tuesday so will test two T-CAN485 then. 

Idle voltage of about 3V is shown on the "Actual" traces above. It was measured directly from the RS485 pins on the lilygo with no bus connected. Similarly the "Expected" trace was taken from the RS485 pins on one of our devices (uses MAX3486 transceiver) with no bus connected. It would be interesting to observe the voltages when the bus is connected between the two devices. I will post the traces of this on Tuesday.

bobemoe, 2024-02-08:
We managed to get one of our original lilygos back from the field which game me something to compare to.

The originals are v1.0 and the new ones we have are v1.1.

I made a small test program outputting on rs495 and uploaded to both, there was a difference between the outputs on both boards, until I set PIN_5V_EN to HIGH.
 
Not sure why this was not needed on the old ones or what it does exactly. Not even 100% sure it was definitely the fix as we compared and changed a lot of things over the coarse of testing, but this was the only difference between the boards we observed.

Anyway the end result is we now have them working!

Is there any change logs for 1.0 to 1.1? Or any docs what PIN_5V_EN high does?

lewisxhe, 2024-02-19:
Hi, @bobemoe 

thanks for your feedback

1. PIN_5V_EN is actually connected to the enable port of the ME2107A50M5G DC booster. It must be set to HIGH to power CAN and RS485.

2. I asked the designer about the difference between Rev1.0 and Rev1.1. Only the input power switching circuit was changed, and other changes were not made.

3. If 1.0 does not require the additional setting of PIN_5V_EN to HIGH, then it may be different from the previous version in that the default on state when using USB power supply is removed (this is just my confirmation based on the previous circuit of the designer who designed this PCB , as this is the case on most boards).

closed 2024-06-24 by Llgok