Xinyuan-LilyGO/T-CAN485 #20: CAN485 ESPHome using CAN

drawlusyk, 2024-04-22:
Does anyone have suggestions on using 2 CAN485 connected vi can h & l.
I have used mcp2515 in esphome successfully but can not get these working. I have them flashed and blinking the led but not send/receive of can messages between the two.
I have wires from h to h and l to l. Is any termination required?

`time:
  - platform: sntp
    on_time:
      - seconds: /5
        then: #trying variations
          - canbus.send:
              data: [0x09, 0x10, 0x11, 0x12, 0x13, 0x14, 0x15, 0x16]
          - canbus.send:
              can_id: 2
              data: [0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08]
          - canbus.send: [ 0x11, 0x22, 0x33 ]    `

`canbus:
  - platform: esp32_can
    tx_pin: GPIO27
    rx_pin: GPIO26
    can_id: 1 #the second device is can_id 2
    bit_rate: 1000kbps #have also tried 250kbps
    on_frame:
    - can_id: 0 #have tried 0 as well as 1/2 for accepting from the other device
      can_id_mask: 0
      use_extended_id: false
      then:
      - lambda: |-
          std::string b(x.begin(), x.end());
          ESP_LOGD("canid 2", "%s", &b[0] );`

teunito, 2024-05-22:
Same here. Can't get CAN working with esphome and T-CAN485. My YAML looks similar to those from @drawlusyk, i'm using interval and not time.
I tried it with a 120R resistor for termination. In https://github.com/Xinyuan-LilyGO/T-CAN485/blob/main/doc/schematic.pdf on page 2 section B-2 there is a 120R resistor (RY2) between CANL and CANH. So i think there is already termination on the board? But i'm not familiar with schema drawings.

I found https://github.com/Xinyuan-LilyGO/T-CAN485/issues/16#issuecomment-1872327090 describing that Output Section is obviously. Is there something similar for CAN?

Anyone any ideas or suggestion?

Llgok, 2024-07-10:
@drawlusyk 
You can refer to the code of other contributors: https://github.com/Xinyuan-LilyGO/T-CAN485/issues/16#issuecomment-2125876729

drawlusyk, 2024-07-11:
> You can refer to the code of other contributors: [#16 (comment)](https://github.com/Xinyuan-LilyGO/T-CAN485/issues/16#issuecomment-2125876729)

Hello. I have uploaded the code used. If this is not what you are referring to, please advise.

-Dennis

mackloo, 2025-02-12:
@drawlusyk Did you manage to get CAN to work with esphome?

drawlusyk, 2025-02-13:
Hello. Yes. I provided details in a previous post. Let me know if you cant find it.

-Dennis

teunito, 2025-02-13:
@mackloo Have a look here: [RS485 ESPHome connect issue #16](https://github.com/Xinyuan-LilyGO/T-CAN485/issues/16#top)

mackloo, 2025-02-13:
Thank you! This is probably it but now my esphome api breaks so I can't connect to it..  will test later. So you have to pull the CAN_SE pin to high if I got it correctly? I checked @drawlusyk example and copied the section about the CAN_SE PIN so that should do the trick right?

mackloo, 2025-02-14:
Solved it! For anyone else that come looking I had to put this into my .yaml file:

output:
  - platform: gpio
    id: CAN_SE
    pin:
      number: 23
      inverted: false # set LOW level

Thanks for pointing me in the right direction!