Xinyuan-LilyGO/T-CAN485 #23: Bluetooth turn on?

sarzyk, 2024-06-19:
Hi I have lilygo T-CAN 485 connected to my inverter Deye. It works very well.
Now I made seccond battery with jk BMS and I want to integrate it with Home Assistant.

JK BMS can connect with my phone via Bluetooth.
Is there any chance (to turn on?) communicate BMS with inverter via Bluetooth?

ThAnks

HBK1, 2024-06-19:
Check this repository: https://github.com/syssi/esphome-jk-bms

sarzyk, 2024-06-24:
Thanks. Is chance tu use the lilygo with deye inverter via modbus and jk BMS via Bluetooth in the same time?

MBRetrofit, 2025-03-30:
> Thanks. Is chance tu use the lilygo with deye inverter via modbus and jk BMS via Bluetooth in the same time?

That's the idea, to connect in the same time the battery (CAN) and the Inverter (RS485).