Xinyuan-LilyGO/T-CAN485 #28: Compilation Error - esp_intr.h no such file or directory

nagup, 2024-07-27:
\Arduino\libraries\ESP32-Arduino-CAN\src\CAN.c:37:10: fatal error: esp_intr.h: No such file or directory
   37 | #include "esp_intr.h"
      |          ^~~~~~~~~~~~
compilation terminated.
exit status 1

Getting above error. Quite vague so was not able to determine what is actually the issue, is the header file too new, too old for this example sketch? 

I have just now tried to rollback the esp32 boards by espressif to version 2.0.11. Then I commented out SD_test(); because not defined. 

I am going to try replacing the initial esp_intr.h with the updated " #include "esp_intr_alloc.h" " as can be found here 

_esp_intr.h: No such file or directory
   37 | #include "esp_intr.h"
      |          ^~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: esp_intr.h: No such file or directory

to overcome that error I added this to the CAN.c file:

```
// #include "esp_intr.h"
#include "esp_intr_alloc.h" // jimn warning depreciated #include "esp_intr.h"
```

_Originally posted by @BEMSman in https://github.com/dalathegreat/Battery-Emulator/issues/116#issuecomment-1865074235__

Llgok, 2024-07-29:
@nagup 
What version of the Arduino esp32 lib are you currently using?

nagup, 2024-07-29:
> @nagup What version of the Arduino esp32 lib are you currently using?

I didn't have the Arduino library installed, only the Espressif one. I just downloaded it, restarted ArduinoIDE and tried to compile but its still failing with the same error. 

I tried another example sketch which came in the zip file, called esp32can_basic, compiling it still fails with same error

Llgok, 2024-07-30:
@nagup 
Arduino libraries of different versions are not compatible with each other. The "main" branch only supports Arduino lib versions below 3.0, while the "arduino-esp32-libs_v3.0.1" branch supports Arduino lib versions 3.0.1 and above. If the wrong version is selected, it will result in compilation errors.
[image]
[image]

Llgok, 2024-07-30:
@nagup 
It looks like you want to configure and run the CAN module on the T-CAN485 device. You should try using the latest branch "arduino-esp32-libs_v3.0.1" and compile the example project [CAN](https://github.com/Xinyuan-LilyGO/T-CAN485/blob/arduino-esp32-libs_v3.0.1/examples/CAN/CAN.ino) with the esp32-libs_v3.0.1 library.
[image]
[image]

nagup, 2024-07-30:
@Llgok 

That worked at last. 

So I deleted the libraries which I pasted into the arduino libraries folder, these were called freenove, modbus and esp32 arduino CAN. 

Replaced them with the libraries in the new branch you showed above. 

Then I restarted arduino and opened an example in the examples folder of the arduino-esp32-libs v3.0.1 archive file and it uploaded fine. 

please note I did not uninstall the arduino and espressif esp32 boards. 

[image]

Llgok, 2024-07-31:
@nagup 
Yes, please use the latest Arduino lib library examples for compilation in the future.

closed 2024-08-21 by Llgok

tejaslithium, 2025-07-06:
Compilation Error - esp_intr.h no such file or directory same again now i am facing