Xinyuan-LilyGO/T-CAN485 #35: 12v power burnt the IC

juliancuni, 2025-01-12:
Is this board 12v capable? 
It was just bad luck that my board happens to be faulty? 

I did programmed it, tested in my desk, usb powered, receiving CAN messages, generatiing PWM to controll a tachometer. 
When I connected to 12v it popped the chip in the picture. 
What is this chip?[image]

andrewfraley, 2025-02-06:
FWIW I was powering mine with 12V no problem for 48h or so.  Did you reverse the polarity maybe?

juliancuni, 2025-02-06:
@andrewfraley I am confident about polarity. Maybe the problem was that I tried to run it in a automotive environment. The voltage readings were some 14v. 
I don't know. Maybe it was bad luck. I'll try the other board and record the whole process. 
Thank you for your feedback anyway, the lilygo guys didn't bother to respond.

juliancuni, 2025-02-06:
@andrewfraley I am confident about polarity. Maybe the problem was that I tried to run it in a automotive environment. The voltage readings were some 14v.
I don't know. Maybe it was bad luck. I'll try the other board and record the whole process.
Thank you for your feedback anyway.

andrewfraley, 2025-02-15:
Mine’s been running for a week now on 12V, but seeing this prompted me to add a 1A inline fuse on the input.

juliancuni, 2025-02-15:
> input

I don't see how a fuse helps with voltage?

andrewfraley, 2025-02-16:
Just in case this thing decides to give up the ghost, mines enclosed with some batteries so I cant risk any sort of fire.

Giddy-Up224, 2025-10-06:
> [@andrewfraley](https://github.com/andrewfraley) I am confident about polarity. Maybe the problem was that I tried to run it in a automotive environment. The voltage readings were some 14v. I don't [...]

I bet that's the issue. 12V in electronics are not always the nominal 12V of automotive circuits.

lewisxhe, 2025-10-07:
We have confirmed with the factory that U6 uses TMI3253, which supports up to 18V input, so 12V is not enough to break down the chip.