Xinyuan-LilyGO/T-CAN485 #37: RS485 and esphome

sobiso, 2025-03-31:
Hi,

I have strange issue, when I connect only my slave device to the bus it doesn't work. But when I also connect to the bus my computer via usb-rs485 the communication between the board and slave works and on computer I can see this communication (request and response)

My esphome configuration
```output:
  - platform: gpio
    id: ENABLE_PIN # Enable the chip
    pin:
      number: GPIO19
      inverted: true
  - platform: gpio
    id: SE_PIN # Enable autodirection
    pin:
      number: GPIO17
      inverted: true
  - platform: gpio
    id: ENABLE_5V_PIN # Enable 5V pin for RS485 chip
    pin:
      number: GPIO16
      inverted: true

uart:
  id: mod_uart
  rx_pin: GPIO21
  tx_pin: GPIO22
  baud_rate: 9600
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: "\n"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);

modbus:
  id: modbus1
  uart_id: mod_uart

modbus_controller:
- id: modbus_device
  address: 0x1   ## address of the Modbus slave device on the bus
  modbus_id: modbus1
  setup_priority: -10
  update_interval: 10s

sensor:
- platform: modbus_controller
  modbus_controller_id: modbus_device
  register_type: holding
  name: "ASP-215 Temerature"
  address: 0x0600    ## address of the register inside the Modbus slave device
  unit_of_measurement: "°C"
  value_type: S_WORD```

lewisxhe, 2025-04-01:
```c
switch:
  - platform: gpio
    id: ENABLE_PIN 
    pin: 
      number: GPIO19
      inverted: true
    restore_mode: ALWAYS_ON
    
  - platform: gpio
    id: SE_PIN
    pin: 
      number: GPIO17
      inverted: true
    restore_mode: ALWAYS_ON
  - platform: gpio
    id: ENABLE_5V_PIN 
    pin: 
      number: GPIO16
      inverted: true
    restore_mode: ALWAYS_ON
```
Try adding Always open

sobiso, 2025-04-02:
I changed resistors 4k7, pull up and pull down, and it works.

closed 2025-04-02 by sobiso

lewisxhe, 2025-04-07:
Which resistor did you change? Can you share?

sobiso, 2025-04-07:
Next to MAX13486, R3 and R8. Both to 330 Ohm